max22-/ESP32-BLE-MIDI #1: Client Scan Crashes ESP32

carloslema, 2020-11-22:
Hi Max,

Thank you so much for putting this library together.

I am looking to connect to the Yamaha BLE MIDI dongle using your library. Unfortunately, it crashes when it is looking for the BLE Device.

The Service UUID and characteristics are correct for the BLE MIDI dongle I am using.

Here is what I sued.
`
#include <Arduino.h>
#include <BLEMidi.h>

void setup() {
    Serial.begin(115200);
    while ( !Serial ) delay(200);
    Serial.println();
    Serial.println("Initializing BLE");

    BLEMidiClient.begin("YamahaBT01"); // Device's Name looking to connect
    // BLEMidiClient.enableDebugging();  // Debugging
}

void loop() {
  int nDevices = BLEMidiClient.scan(); // CRASHES ESP32
  if(nDevices > 0) {
    Serial.print("Devices ");
    Serial.println(nDevices);
  }
}`

I also looked at Neil's code that tries to create a server, not a client — it crashes too.

My goal is to send program change and cc control over BLE to the Yamaha dongle.

Any help is greatly appreciated.

best,
\\ carlos

max22-, 2020-11-22:
Hi Carlos,
I am happy to see that it is useful for you !
I think the error is in this line :
BLEMidiClient.begin("YamahaBT01");
In fact, the parameter you give to the "begin" function is the name you want to give to your ESP, not the name of the device you want to connect to. My example wasn't very clear.
Can you try to uncomment the debugging line just below and send me the output ?

best regards
Maxime

carloslema, 2020-11-22:
Hi Maxime, Thanks for your response.

Yeah, I tried  BLEMidiClient.begin("ESP32-BLEMIDI") before, and uncommenting the debugger — crickets.
Even with verbose debugging the console just prints, "Initializing BLE" — but nothing else.

BTW, I am using the Adafruit Huzzah ESP32.

Also, I have been using another script, sans a nice class, that connects and sends MIDI to the BT01 dongle — but packaging multiple MIDI packets is where I am running into issues —— and that's where I started to look for a better alternative, possibly yours.

best,
\\ carlos

max22-, 2020-11-22:
Do you use Arduino IDE or PlatformIO ?
Can you send me the full log of the serial console ?

carloslema, 2020-11-23:
Hi Maxime,
Hope all is well with you and yours.
I use both. Lately, Atom with PlatformIO has been a little weird. I've been using Arduino Beta 1.9.0 on Mac Catalina for anything IoT — the Arduino IDE is terrible but works for now. I don't want to use PlatformIO in VScode since it messes up my setup for Google APIs, Heroku, Node, Vue, Mongo etc. Let me try your library within the  Atom/PlatformIO IDE to see if makes a difference—it shouldn't.
That's what's puzzling, with your library, with enableDebugging(); uncommented  I don't get a log — the console is printing other messages, but nothing from BLEMidi. Last night I got my other code to package and send midi correctly.
best, \\ carlos 

— helping brands create memorable experiences with their customers

+1 212.472.4631
carlos_lema@yahoo.com
http://carlos-lema.com

------
LinkedIn: http://linkd.in/h0KrIT

max22-, 2020-11-23:
Hi Carlos,
I am interested by the other messages, can you send me the complete log you get in the serial monitor ?

georaura, 2020-12-24:
Hi Carlos, 
I have the same problem. I resolve this, changing the BLEMidiClient.cpp file: 

// I have an error on this line (null pointer, i think):
  if(device.getServiceUUID().equals(BLEUUID(MIDI_SERVICE_UUID))) {

// Then I change for this code:

 if (device.haveServiceUUID() && device.isAdvertisingService(BLEUUID(MIDI_SERVICE_UUID))) {	  
 
Please comment if you resolve your problem..

max22-, 2020-12-24:
Hi georaura,
thanks for your feedback, and merry christmas !
Can you tell me what esp32 board you are using, and what midi device ?

carloslema, 2020-12-25:
@georaura Thanks for the fix — Yes, indeed this works on the Adafruit HUZZAH ESP32.

georaura, 2020-12-27:
Hi Maxime. 
Merry chritsmas too !!
I'm using an HiLetgo ESP32 dev module to control an Avid Eleven Rack guitar effects connected through a midi BLE dongle (Doremidi model from Shenzhen Huashi Technology Company) . I send program change messages and with the code fix it works fine..
Thanks for the library...

carloslema, 2020-12-27:
@georaura I am going old school to control a TriAxis/2:90, Cmaj & Loop Switch I made. I am expanding forntoh/LcdMenu — although the 20x4 LCD is very limiting.

[image]

max22-, 2020-12-30:
Hi everyone,
I have integrated the bug fix from @georaura into the library, and i have also implemented timestamps.
i have made a small release this morning, it is available in Platformio and Arduino IDE.
I will mark it as closed, but don't hesitate to contact me if you find some other bugs.
And happy new year !

closed 2020-12-30 by max22-