max22-/ESP32-BLE-MIDI #2: Not receiving simultaneous events/polyphony

CesarBrz, 2021-03-06:
My system:
Esp32
Macbook Pro late 2015 15" 

Issue happens when sending quantized midi chords from a DAW (tried with Reaper 6 and Pro Tools First) .

If the DAW send a 3 note chord, with the 3 notes starting at the exact same tick, Esp32 will recognize only 1 of the 3 note on events (the first sent from the DAW, I suppose).

The same thing applies both to simultaneous note off events or mixed note on and off events (for example, a note off event on one note happening at the same time as a note off event of other note).

This may cause stuck notes or avoid some note(s) being triggered since if a note on events is received simultaneously with a note off event, only one of them is received by ESP32.

Summing up, when multiple events are sent by the DAW simultaneously, only one is recognized and the others aren't received/recognized by ESP32.

My specific project is building a bluetooth midi receiver to control some synthetizers through my DAW wirelessly. My workaround for now is to slightly shift the midi notes so there's never 2 events happening at the same time.

max22-, 2021-03-06:
Hi Cesar,
I am aware of this bug. I have already observed it too. It is caused by "running status messages", i.e. when several messages are packed because they contain the same command on the same channel.
I will work on it as soon as I can, it shouldn't be difficult to correct the receiving code.
best regards,
Maxime

CesarBrz, 2021-03-07:
Hi Maxime,
First and foremost, thank you for the library and the reply.

While working with another project (a "VU meter" for a Roland Vs2400, this mixer sends the volume of the tracks as the velocity of the notes 0-23 of channel 16) I have faced a similar situation, having to "filter" the incoming data to get only the relevant bytes.

By your reply, it seems the issue is being caused by multiple note on/off data following a single note on/off command, right?
If so, maybe the following code from my project may help.

void MidiHandler ()
{
    byte BufferIn[3]; //input buffer
   
    //Filter phase

    BufferCheck[2] = BufferCheck [1];
    BufferCheck[1] = BufferCheck [0];
    BufferCheck[0] = Serial1.read();  

    //If not valid volume track data, abort - loop will call this function
    if (BufferCheck[2] != 248 || BufferCheck[1] != 175 || BufferCheck[0] > 23)  {return;}

    //End of Filter found the 3 bytes that signal the start of the chain of track volume data
    // 248 = Midi Tick / 175 = Note On on Ch 16 / 0-23 = Tracks 1 to 24

    BufferIn[0] = BufferCheck[0]; //stores the first mixer track number on the first array byte
    BufferIn[1] = Serial1.read(); //reads velocity for the first mixer track in the array

    //While data is valid and the Track number is lower than the number of tracks on the led strip
    //While data is consistent BufferIn[0] have the note number, NumTracks is the criteria to determine if data is valid
    while(BufferIn[0] < NumTracks)
    {
      //Proceed to process the incoming data
      //LightenTrack (BufferIn[0], BufferIn[1], LedsPerTrack);
      
      BufferIn[0] = Serial1.read(); //stores new note
      BufferIn[1] = Serial1.read(); //stores new velocity
    }

  //loop break => no channel note. Goes Back to filter phase
}

max22-, 2021-03-13:
Hi Cesar,
I have made a new release of the library, this bug should now be corrected. It is not yet available in the Arduino IDE (but it should be in a couple hours). But you can find it here or in PlatformIO. The new version is 0.2.0
Could you tell me if everything works for you ?
If you want, I have also made a new example, which dumps all the messages that the ESP32 receives, to make some quick tests.

CesarBrz, 2021-04-04:
Just tested it and new it seems to be working just fine!

closed 2021-04-04 by CesarBrz