max22-/ESP32-BLE-MIDI #3: No succesful connection in Client mode

RobertoHE, 2021-04-06:
Hi Max22.

I try to connect a ESP32 pico in client mode to a Roland AX-edge (a MIDI keyboard with syntetizer and Bluetooth LE v4.1) like server. 

I run the example code of clientBLE and ESP32 shows by Serial (debug enabled) that it is connected to AX-Edge but AX-Edge does not notify any comunication with ESP32. I can notice that because when AX-edge has a succesful connection, the Bluetooth icon changes into black background, and this does not happen when I try to connect with ESP32. I haven't had any problem of connection when I try to connect to AX-Edge with tablets, smartphones or pc windows, only with ESP32 and this library.

If I modify slightly the example code by adding a noteOn callback and I try to send any note from AX-edge to ESP32, there is no response. Only callbaks of connect and disconnect events work. 

I checked the UUID services and characteristics of this library and they are OK with Midi BLE definiction and they are the same as AX-Edge UUID. 

I have doubts whether the library really notifies its characterists to an external device or something else regarding notificating back to the external device to acomplish the comunication.

I cannot offer you any debug from AX-edge because I cannot take any log or debug feedback from the device.

Can you help me please?

Thank you for your help and for the library.

RobertoHE, 2021-04-14:
Hi again.

I have tested some example codes for ESP32 and BLE and I could connect ESP32 to my AX-edge using other BLE base library. 

My connection problems was related with Authentificacion (encrypt connection) callbacks, that BLE base lib didnt work correctly.

I only can connect  the ESP32 module correctly using nimBLE lib. In other cases using other lib like BLEDevice or similar libraries based on esspresif lib was imposible run BLE callbacks correctly, and those callbacks are necessary for implement a secure comunication, bond, authentication method or recive any subcribed characterics.

I adapted client example code of nimBle with a handler of Midi.h and Debug.h class of this repo and it is working perfectly. I havent remade full lib yet for implent sendPackect, but it is functional for reciving midi menssages from AX-Edge.

It may be interesting to change the communication base library to nimBLE because it is equal to or more powerful than the basic library, gives less problems and takes up half resources.

Best Regards

jcgfsantos, 2021-04-29:
Hi @RobertoHE ,

I have the same problem as you when trying to connect to a Roland FP-30 with this library. When I run it on a ESP32 it says that is is connected and sending notes but no sound comes out. I'm able to pair and send BLE commands to play sounds using Bluetooth LE explorer in windows 10.

Would it be possible for you to share the modified codes?

Thanks!

max22-, 2021-05-04:
Hi @RobertoHE and @jcgfsantos , 
sorry for replying late but i was a little busy lately.
it is a bit difficult for me to test the client mode, because I have only one midi device (my little Nux guitar amp).
Thanks @RobertoHE for pointing to nimBle. It looks quite nice. I will try to see what I can do with it.
But if you want, feel free to share your code, you will be given credit for your work.
Best regards

RobertoHE, 2021-05-04:
Hi @max22-. 
I will try to code the lib with nimBLE this month, if I have time. If I have any advance I will try to fork the branch.

Hi @jcgfsantos my test code only recive notes or other midi data, but It cant send any data yet. If I want to send any Midi data, I need to develop a class method in the lib that indicate how the class send info over BLE. In other words, I have to rewrite the sending funtions ussing nimBLE. Let me some time for do that.

RobertoHE, 2021-05-21:
Hi @jcgfsantos 

I changed BLE lib to nimBLE in my fork.

https://github.com/RobertoHE/ESP32-BLE-MIDI.git

I tested the example 02-Base-Midi-Client. Enable debug line for view incoming data from server or add callbacks from example 03-Receving-Data.

Please, test it and tell me if it works in your device.

Regards

jcgfsantos, 2021-05-22:
Hi @RobertoHE,

I had to change in BLEMidiBase.h, line 4

from:

#include <nimBLEDevice.h>
to 
#include <NimBLEDevice.h>

to be able to compile.

It is able to connect to the piano (I see the blue light) but I don't hear any note. 
Everything looks OK so I don't know why it is not working. If I have some time I will give it a closer look.
Meanwhile I was also able to use NimBLE to connect to the piano and do what I had in mind (I was also able to send a midi message to play a note but I was more interested in sending a Sysex message).
You may find my code in https://github.com/jcgfsantos/BLEMidiMETROland

Thanks!

RobertoHE, 2021-05-23:
Hi again @jcgfsantos 

I think that I fixed sending messages. NimBLE writeValue needed one more parametrer than native BLE lib.

Would you test the change, please?

jcgfsantos, 2021-05-23:
Hi @RobertoHE 

It is working! Very nice work.

Thanks