max22-/ESP32-BLE-MIDI #5: Callbacks are never called

buzz-dk, 2021-09-16:
Hello Maxime.
Using your example code 04-Dump-Messages.ino with an ESP32 az-delivery-devkit-v4 board I found the following:
1: Callbacks for connect and disconnect were never called.
2: The board freezes on disconnect, ie it refuses to establish another connection.

I have fixed my problem by defining the callbacks before calling .begin()
I also added a restart command tho the disconnect callback. At least it can be used as a temporary solution.

I hope you find this useful. Thanks for sharing your code.
Example shown below.
Best regards
Kim

void setup() {
  Serial.begin(115200);
  BLEMidiServer.setOnConnectCallback([]() {
    Serial.println("Connected");
  });
  BLEMidiServer.setOnDisconnectCallback([]() {
    Serial.println("Disconnected");
    ESP.restart();
  });
  BLEMidiServer.begin("MIDI device");
  BLEMidiServer.setNoteOnCallback(onNoteOn);
  ........

max22-, 2021-09-20:
Hi Kim,
thank you very much for spotting these bugs, and the information you provided !
I have corrected them, and published the code on the arduino library manager and platformio.
Now the example works without needing to modify it, and we can connect and disconnect as much as we want.
Can you tell me if it works also for you ?
Best regards,
Maxime

buzz-dk, 2021-09-20:
Hi Maxime
I tested your solution 0.2.2 with the ESP32 az-delivery-devkit-v4 on PlatformIO, and I can confirm that It now functions as expected:
1: Callbacks for connect and disconnect are called even when defined later than the .begin() call
2: A disconnect do no longer freeze the board. 
Many thanks!!
Cheers
Kim

max22-, 2021-09-20:
Ok cool !
cheers,
Maxime

closed 2021-09-20 by max22-