max22-/ESP32-BLE-MIDI #20: Request example

DRCRecoveryData, 2023-03-11:
Hi,

Could you create example code for one potentiometer as 10 bits or 14 bits for esp32-s3 devkit c?

max22-, 2023-03-11:
hi, what kind of message do you want to send ? pitch bend ?
(other types of messages are limited to 7 bits)

DRCRecoveryData, 2023-03-12:
> hi, what kind of message do you want to send ? pitch bend ?
> (other types of messages are limited to 7 bits)
> 
> 
> > Hi,
> >
> > Could you create example code for one potentiometer as 10 bits or 14 bits
> > for esp32-s3 devkit c?

I use to mapping to ableton live with knob and volume faders, i have plan to made 16-32 potentiometers because esp32-s3 run faster than atmega and stm32 chip

max22-, 2023-03-22:
I sent a reply 10 days ago by mail, so i post it here :

> Hi,

>i've made a small example, it will arrive soon in the arduino IDE if you update the library, or you can find it here on github :

 >https://github.com/max22-/ESP32-BLE-MIDI/blob/master/examples/05-Simple-Knob/05-Simple-Knob.ino

> I don't have ableton, i've tested it with LMMS, it works. Good luck :)