max22-/ESP32-BLE-MIDI #21: automatically disconnects after 5 seconds

Ash-86, 2023-03-12:
Running the basic midi device example connects for only 5 seconds, then automatically disconnects. 

running windows 11.

efraser67, 2023-05-10:
I'm having the same issue, connects and sends 3 or 4 messages then disconnects and won't reconnect without re-pairing the ESP32S3.

efraser67, 2023-05-11:
Found a work around for windows 11.  Installing the KORG-BLE midi driver and setting the Bluetooth Helper service to Automatic seems to allow a stable connection with the ESP32S3.  Shows up as a midi IN without any need for additional software like MIDIBerry or loopMidi etc...

Ash-86, 2023-05-11:
Ah great! I had found a solution using loopMidi and Midi Berry however this worked only when using the Arduino-BLE-MIDI by lathoub (https://github.com/lathoub/Arduino-BLE-MIDI) and didn't seem to work with the ESP32-BLE-MIDI by Max22.

good to know that there is a simpler solution with KORG-BLE.

FixItWizard, 2023-12-18:
I'm still having disconnection issues with my Windows 10 using the lathoub's BLE-MIDI on my ESP32 WROOM after installing the KORG-BLE driver. This is so frustrating. I believe my laptop has Bluetooth 5.2 or 3. Not sure if that's the issue or not.

h4nsen, 2024-02-21:
Can I ask how your devices appear in Windows?

I'm on Windows 10 and are having a lot of problems getting my esp32 to work at all with Windows. No problems with my iOS device. I've installed KORG BLE-MIDI Device driver (the r4-version that is suppose to fix issues introduced by an update to Windows)

But my device still just shows under Bluetooth - no device under Sound, video and game controllers other than the KORG-device.

[image]

Do you manually choose a driver for your device?