max22-/ESP32-BLE-MIDI #22: No application detecting the device

Bugsia, 2023-05-26:
Hello,

I can´t get any application to recieve the signals, that I send, except for MIDIOX but that software is only used to check if any signals are recieved. But softwares like MIDIberry don´t see the input at all, while softwares like Voicemeeter and MIDI Mixer see the inputs but don´t recieve the send signals.
What am I doing wrong?

`#include <Adafruit_MCP23X17.h>
#include <Wire.h>
#include <SPI.h>
#include <driver/adc.h>
#include <BLEMidi.h>
#include "Convert.h"

#define ADDRESS 0x20
#define REGISTER_A 0x00
#define REGISTER_B 0x01
#define OUTPUT_SET 0x00
#define BANK_A 0x12
#define BANK_B 0x13

#define SIZE 8
#define POTS_SIZE 4
#define CONTROL_CHANGE_RANGE 127
#define SENSI_POTS 15

Convert converter;
Adafruit_MCP23X17 mcp;

String ledAdr[SIZE] = {"B7", "A0", "B5", "B4", "B3", "B2", "B1", "B0"};
String btnAdr[SIZE] = {"B6", "A1", "A2", "A3", "A4", "A5", "A6", "A7"};

int leds[SIZE] = {0,0,0,0,0,0,0,0};
int buttonsOld[SIZE] = {0,0,0,0,0,0,0,0};
int buttons[SIZE] = {0,0,0,0,0,0,0,0};
boolean btnToggle[SIZE] = {true, true, true, true, false, false, false, false};

int sda_pin = 7;
int scl_pin = 6;

int[POTS_SIZE] oldValue = {0,0,0,0};
int[POTS_SIZE] value = {0,0,0,0};
int[POTS_SIZE] maxPotValue = {3500, 3500, 3500, 3500};

void setup() {
  /**
   * Setup for Serial line
   */
  Serial.begin(115200);
  Serial.println("ESP32 MIDI Controller");

  /**
   * Setup for GPIO Expander
   */
  Wire.setPins(sda_pin, scl_pin);
  Wire.begin();
  if(!mcp.begin_I2C(0x20, &Wire)) {
    Serial.println("Error in connecting with GPIO Expander");
  }

  //Configure Pull Ups(Input)
  for(int i = 0; i < SIZE; i++) {
    int pos = calculateMcpPos(btnAdr[i]);
    mcp.pinMode(pos, INPUT_PULLUP);
  }

  //Configure Pull Downs(Output) and Turn them off
  for(int i = 0; i < SIZE; i++) {
    int pos = calculateMcpPos(ledAdr[i]);
    mcp.pinMode(pos, OUTPUT);
    mcp.digitalWrite(pos, 0);
  }

  /**
   * Setup for ADC
   */
  adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_DB_6);
  adc1_config_channel_atten(ADC1_CHANNEL_1, ADC_ATTEN_DB_6);
  adc1_config_channel_atten(ADC1_CHANNEL_2, ADC_ATTEN_DB_6);
  adc1_config_channel_atten(ADC1_CHANNEL_3, ADC_ATTEN_DB_6);

  /**
   * Setup for MIDI over Bluetooth
   */
  Serial.println("Initializing bluetooth");
  BLEMidiServer.begin("Basic MIDI device");
  Serial.println("Waiting for connections...");
  /*
  while(!BLEMidiServer.isConnected()) {
    Serial.println("Waiting for connections...");
    delay(1000);
  }
  Serial.println("Connected!");
  */
}
 
void loop() {
  if(BLEMidiServer.isConnected()) {
      midiChecker();
  }
  
  updater();
  checker();
}

void updater() {
  for(int i = 0; i < SIZE; i++) {
    buttonsOld[i] = buttons[i];
  }
  
  updateButtonStates();
  updateLedStates();
}

void checker() {
  checkButtonsAndUpdateLeds();
}

void midiChecker() {
  checkPotsAndSendMIDI();
}

/**
 * BUTTON LOGIC
 */
void checkButtonsAndUpdateLeds() {
  for(int i = 0; i < SIZE; i++) {
    boolean released = buttonReleased(i);
    boolean pressed = buttonPressed(i);
    boolean toggleBtn = btnToggle[i];

    if(toggleBtn && released) {
      toggleLed(i);
    }
    else if(!toggleBtn && released) {
      turnLedOff(i);
    }
    else if(!toggleBtn && pressed) {
      turnLedOn(i);
    }
  }
}

boolean buttonReleased(int num) {
  if(buttonsOld[num] == 1 && buttons[num] == 0) {
    return true;
  }
  else {
    return false;
  }
}

boolean buttonPressed(int num) {
  if(buttonsOld[num] == 0 && buttons[num] == 1) {
    return true;
  }
  else {
    return false;
  }
}
 
void updateButtonStates() {
  for(int i = 0; i < SIZE; i++) {
    String curAdr = btnAdr[i];
    int pos = calculateMcpPos(curAdr);;

    if(!mcp.digitalRead(pos)) {
      buttons[i] = 1;
    }
    else {
      buttons[i] = 0;
    }
  }
}

/**
 * LED LOGIC
 */
void toggleLed(int num) {
  int cur = leds[num];
  if(cur == 0) {
    leds[num] = 1;
    sendMIDIon(num);
  }
  else {
    leds[num] = 0;
    sendMIDIoff(num);
  }

  updateLedStates();
}

void turnLedOn(int num) {
  leds[num] = 1;
  sendMIDIon(num);
  updateLedStates();
}

void turnLedOff(int num) {
  leds[num] = 0;
  sendMIDIoff(num);
  updateLedStates();
}

void updateLedStates() {
  for(int i = 0; i < SIZE; i++) {
    String curAdr = ledAdr[i];
    int pos = calculateMcpPos(curAdr);

    mcp.digitalWrite(pos, leds[i]);
  }
}

/**
 * MIDI LOGIC
 */
void checkPotsAndSendMIDI() {
  //Update Values
  for(int i = 0; i < SIZE_POTS; i++) {
    oldValue[i] = value[i];
  }
  value[0] = adc1_get_raw(ADC1_CHANNEL_0);
  value[1] = adc1_get_raw(ADC1_CHANNEL_1);
  value[2] = adc1_get_raw(ADC1_CHANNEL_2);
  value[3] = adc1_get_raw(ADC1_CHANNEL_3);

  //Check for changes and send MIDI accordingly
  for(int i = 0; i < SIZE_POTS; i++) {
    if((oldValue[i] - value[i]) > SENSI_POTS) {
      sendMIDIPots(i);
    }
  }
}

void sendMIDIPots(int num) {
  if(BLEMidiServer.isConnected()) {
    Serial.println("SENDED CONTROL");
    BLEMidiServer.controlChange(0, (77 + num), mapPotValueToMIDI(value[i]));
  }
}

double mapPotValueToMIDI(int curValue) {
  double divider = (double)CONTROL_CHANGE_RANGE / (double)curValue;
  return (double)curValue * divider;
}

void sendMIDIon(int num) {
  if(BLEMidiServer.isConnected()) {
    BLEMidiServer.noteOn(0, (69 + num), 127);
  }
}

void sendMIDIoff(int num) {
  if(BLEMidiServer.isConnected()) {
    BLEMidiServer.noteOff(0, (69 + num), 127);
  }
}

/**
 * OTHER LOGIC
 */
int calculateMcpPos(String adr) {
  int pos = 0;
  
  if(adr.charAt(0) == 'B') {
      pos += 8;
  }
  pos += adr.charAt(1) - '0';;

  return pos;
}

String generateBinaryFromLedStates() {
  String binary = "";
  for(int i = 0; i < SIZE; i++) {
    binary += String(leds[i]);
  }

  return binary;
}

int binaryToHex(char *binary) {
  int decimal = converter.binaryToDecimal(binary);
  String hex = converter.decimalToHexa(decimal);
  return hex.toInt();
}`

max22-, 2023-05-27:
Hello,
if midiox can see your midi messages, it is likely a problem with windows or your other software.
i can't help you that much because i don't have windows on my PC and i've never used those software.

Bugsia, 2023-05-28:
Okay, thank you! I will try it on a fresh install of Windows

closed 2023-05-28 by Bugsia