max22-/ESP32-BLE-MIDI #23: Pitchbend

marlow77, 2023-06-18:
Hello!

Thank you for making this library and the examples are great!

I can't explain the feeling of accomplishment I got just from making it okay the midi key over to my Android phone, super cool!

Can you help me figure out how I would send pitchbend messages?  I tried multiple ways with the header information files, but I couldn't get it to work.

Thanks in advance,

Marlow

max22-, 2023-06-19:
Hello, I've replied by mail but i don't see it here so i post it here just in case :

 ```
Hello,

thank you for your message !

for the pitch bend you should use this function :

https://github.com/max22-/ESP32-BLE-MIDI/blob/master/src/utility/Midi.h#L21

for example, if you want to simulate a modulation wheel with a potentiometer connected to pin 39 you could do something like this in your main loop :

float semitones = (float)analogRead(39) / 4096.0 * 4 - 2;

BLEMidiServer.pitchBend(0, semitones);

In general pitchbend goes from -2 to +2 semitones, but if your android app doesn't have the same range, you can provide it as a third argument.

I hope this can be helpful for you.

Maxime
```

marlow77, 2023-06-20:
Thank you for getting back to me so quickly!  I'll report back how it goes.  Thanks!Marlow

https://github.com/max22-/ESP32-BLE-MIDI/blob/master/src/utility/Midi.h#L21

closed 2023-06-27 by max22-