max22-/ESP32-BLE-MIDI #26: Recent MacOS & iOS won't discover BLE device

lomoalapimienta, 2023-12-09:
Hi there!
Have been playing a lot with this library but recently found that recent MacOS and iOS devices won't discover the ESP32 BLE device at all. The same code works perfect on earlier MacOS versions (ie: High Sierra) but won't even discover the Bluetooth device in the list in Ventura. Same with an iPhone 14. On the other hand, all the Android and Windows devices tested so far work flawlessly.
Doing some research, seems to be related to a security issue, as Apple might have included a new requirement, not allowing BLE devices without encryption/authentication schemes. This is the only hint I have.
Can you help me understand if it is possible to configure the BLE connection with a PIN and/or other security parameters?
Thanks!

LekoPhantom, 2024-03-29:
I too am seeing this behavior. I am not sure how much help I can be, but if I get a chance, I will take a look. I do know that the [SparkIO](https://github.com/paulhamsh/Spark) project does work with the iPad, so their bluetooth stack may be more amenable to the newer apple security protocols.

lomoalapimienta, 2024-03-29:
> I too am seeing this behavior. I am not sure how much help I can be, but if I get a chance, I will take a look. I do know that the [SparkIO](https://github.com/paulhamsh/Spark) project does work with the iPad, so their bluetooth stack may be more amenable to the newer apple security protocols.

Hi @LekoPhantom it seems that BLE devices not advertising encryption capabilities won't be displayed and available to connect using the Bluetooth menu. But I've finally managed to connect the device using the Audio MIDI setup application provided by the Mac OS. 
Give it a try and let me know!

LekoPhantom, 2024-03-29:
I do not have a current mac, but I did have success with the same workaround on ios. I found the app via [this issue](https://github.com/tttapa/Control-Surface/issues/101#issuecomment-579675788) for [Control Surface ](https://github.com/tttapa/Control-Surface. I used the [BLEScanner app](https://apps.apple.com/us/app/ble-scanner-4-0/id1221763603)

lomoalapimienta, 2024-03-29:
The MIDI setup app has been available for many years. I am using also a
2013 MacBook pro with high Sierra. This OS shows the device in Bluetooth
menu

ezequielabregu, 2025-03-31:
You solve it by using an app called "Conji" (MacOS / IOS)