max22-/ESP32-BLE-MIDI #27: make BLEMidiServer.sendMessage() public

SunboX, 2023-12-20:
Please make the `BLEMidiServer.sendMessage()` method public, so we can send messages like this:

```c++
uint8_t midiMessage[] = {
    0xF0, // sysex
    0x00,
    0x21,
    0x3C,
    0x00,
    0x10,
    0x33,
    0x60,
    0x5A,
    0x00,
    0x7F,
    0xF7 // end of sysex
};
BLEMidiServer.sendMessage(midiMessage, sizeof(midiMessage));
```

max22-, 2024-01-02:
Hi, sorry for replying late.
it could be nice to directly have a sysex function, what do you think ? would you be interested to make a pull request ?

ellisgl, 2025-03-18:
I'll 3rd that!

SunboX, 2025-04-04:
> Hi, sorry for replying late. it could be nice to directly have a sysex function, what do you think ? would you be interested to make a pull request ?

Hi, sorry that I'm also late. ;) SysEx would be nice, but in general it would be great to have the sendMessage method public to be able to send non standard MIDI messages.