max22-/ESP32-BLE-MIDI #31: Crashing after update

ChrisGleich1AufsMaul, 2024-06-09:
After I updated bordmanager and library, my ESP32S3 keeps crashing when starting the server („ BLEMidiServer.begin("Basic MIDI device");“).

max22-, 2024-06-09:
What message do you receive ?

ChrisGleich1AufsMaul, 2024-06-09:
Sure, that's what I get:

Backtrace: 0x403832f3:0x3fcf0bc0 0x40381c75:0x3fcf0bf0 0x40383680:0x3fcf0c20 0x40383676:0xa5a5a5a5 |<-CORRUPTED

ELF file SHA256: d718bbb47b52a13d

Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376b65
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
Initializing bluetooth
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (ipc0) 
Core  0 register dump:
PC      : 0x403832f6  PS      : 0x00060036  A0      : 0x80381c78  A1      : 0x3fcf0bc0  
A2      : 0x3fc98394  A3      : 0xffffffff  A4      : 0x00060023  A5      : 0x00060023  
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x80383489  A9      : 0x3fc99bd0  
A10     : 0x00060023  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x00000001  
A14     : 0x03c98290  A15     : 0x3fc98290  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

max22-, 2024-06-09:
can you send me your code and .hex file ?

ChrisGleich1AufsMaul, 2024-06-09:
Well, this happens with the lib example (eg. 01-Basic-Midi_Device.ino)
Which .hex file do you need?

max22-, 2024-06-09:
The one produced when you compile the program. do you use Arduino IDE or PlatformIO ?

ChrisGleich1AufsMaul, 2024-06-09:
I'm using Arduino IDE 2.3.2
When I downgrade the board manager to 2.0.17 it works again

max22-, 2024-06-09:
Same error for me.
Here is what i get when i decode the backtrace :
`
$ ~/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-addr2line -pfiaC -e /tmp/arduino/sketches/48CC8EBC2FE0039500F6F88C86C42DBB/01-Basic-Midi-Device.ino.elf 0x40381c97:0x3fcf0bc0 0x40380619:0x3fcf0bf0 0x40382024:0x3fcf0c20 0x4038201a:0xa5a5a5a5

0x40381c97: xPortSetInterruptMaskFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:560
 (inlined by) xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:500
0x40380619: vPortEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:575
 (inlined by) vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3793
0x40382024: _frxt_dispatch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:450
0x4038201a: _frxt_int_exit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:245
`

Unfortunately it will be difficult to find the cause of this bug, as it happens inside the esp-idf framework, and i am currently busy on other projects so i don't have that much time to work on it.