max22-/ESP32-BLE-MIDI #35: Disable NimBLE debugging in serial monitor

OSCmixer, 2024-10-17:
Hello, I tried using BLEMidiServer.disableDebugging();, but nimBLE is still showing in the serial monitor. I have the latest version of the Arduino library, version 0.3.2, installed.

Should protected void receivePacket(uint8_t *packet, uint8_t packetSize); be in midi.h? Can you share an example of how to create a new task, for instance, in CPU 1 with priority 24? GPT chat attempted to handle MIDI and process MIDI, but nothing worked. :D I have now modified the original library, but it would be better not to have to modify it in case a new update comes.

max22-, 2024-10-17:
can you show me your code ?
you can create a task with [xTaskCreatePinnedToCore](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/freertos_additions.html)

OSCmixer, 2024-10-17:
cpp_freertos::Task midiTaskNOTES("MIDITaskNOTES", 3024, 24, []() { // Task priority 24
    uint8_t data[3]; // Buffer for Note On/Off messages (3 bytes)
    size_t size = sizeof(data); // Size of the buffer
    const uint32_t noteInterval = 2; // Time interval for processing notes (2 ms)
    static uint32_t lastNoteProcessTime = 0;

    while (true) {
        uint32_t currentMillis = millis();

        // Check if it's time to process MIDI notes
        if (currentMillis - lastNoteProcessTime >= noteInterval) {
            // Receive MIDI Note messages
            BLEMidiServer.receivePacket(data, size); // Check if size needs adjusting based on received message

            if (data[0] >= 0x80) { // Check for valid MIDI status byte
                switch (data[0] & 0xF0) { // Filter out status byte
                    case 0x90: // Note On
                        onNoteOn(data[0] & 0x0F, data[1], data[2], xTaskGetTickCount()); // Channel, Note, Velocity, Timestamp
                        break;
                    case 0x80: // Note Off
                        onNoteOff(data[0] & 0x0F, data[1], data[2], xTaskGetTickCount()); // Channel, Note, Velocity, Timestamp
                        break;
                }
            }

            // Update last process time
            lastNoteProcessTime = currentMillis;
        }

OSCmixer, 2024-10-17:
#include <Arduino.h>      // Arduino library
#include <BLEMidi.h>      // ESP32-BLE-MIDI by Maxime ANDRÉ
#include "freertos-all.h" // Addon by Phil Schatzmann, https://github.com/pschatzmann/arduino-freertos-addons

//debug freeRTOS and BLEmidi
#define DEBUG_MIDI_TASK  // Comment this line to disable debugging for performance mode

// MIDI NOTE ON-OFF Channel 1
void onNoteOn(uint8_t channel, uint8_t note, uint8_t velocity, uint16_t timestamp) {
    if (channel == 0) { // MIDI channel 1
      Serial.printf("Received note on: channel %d, note %d, velocity %d (timestamp %dms)\n", channel, note, velocity, timestamp);
    }
}

void onNoteOff(uint8_t channel, uint8_t note, uint8_t velocity, uint16_t timestamp) {
    if (channel == 0) { // MIDI channel 1
        Serial.printf("Received note off: channel %d, note %d, velocity %d (timestamp %dms)\n", channel, note, velocity, timestamp);
    }
}

void connected() {
    Serial.println("A device has connected.");
}

void onControlChange(uint8_t channel, uint8_t control, uint8_t value) {
    Serial.printf("Control change: channel %d, control %d, value %d\n", channel, control, value);
}

// MIDI TASK for notes, experiment
cpp_freertos::Task midiTaskNOTES("MIDITaskNOTES", 3024, 24, []() { // Task priority 24
    uint8_t data[3]; // Buffer for Note On/Off messages (3 bytes)
    size_t size = sizeof(data); // Size of the buffer
    const uint32_t noteInterval = 2; // Time interval for processing notes (2 ms)
    static uint32_t lastNoteProcessTime = 0;

    while (true) {
        uint32_t currentMillis = millis();

        // Check if it's time to process MIDI notes
        if (currentMillis - lastNoteProcessTime >= noteInterval) {
            // Receive MIDI Note messages
            BLEMidiServer.receivePacket(data, size); // Check if size needs adjusting based on received message

            if (data[0] >= 0x80) { // Check for valid MIDI status byte
                switch (data[0] & 0xF0) { // Filter out status byte
                    case 0x90: // Note On
                        onNoteOn(data[0] & 0x0F, data[1], data[2], xTaskGetTickCount()); // Channel, Note, Velocity, Timestamp
                        break;
                    case 0x80: // Note Off
                        onNoteOff(data[0] & 0x0F, data[1], data[2], xTaskGetTickCount()); // Channel, Note, Velocity, Timestamp
                        break;
                }
            }

            // Update last process time
            lastNoteProcessTime = currentMillis;
        }

        // Monitor stack usage for debugging
        #ifdef DEBUG_MIDI_TASK
        UBaseType_t highWaterMark = uxTaskGetStackHighWaterMark(NULL); // Get current task's stack usage
        Serial.print("Remaining stack for midiTaskNOTES: ");
        Serial.println(highWaterMark); // Print how much of the stack remains (in words)
        #endif

        // Small delay to reduce CPU usage
        vTaskDelay(1 / portTICK_PERIOD_MS); // Adjust delay as needed for responsiveness
    }
});

void setup() {
    Serial.begin(115200); // Initialize serial communication
    // Initialize BLE MIDI server
    BLEMidiServer.begin("BLE MIDI");
    BLEMidiServer.setOnConnectCallback(connected);
    BLEMidiServer.setOnDisconnectCallback([]() {
        // Serial.println("Disconnected"); // Uncomment for debugging
    });
    BLEMidiServer.setNoteOnCallback(onNoteOn);
    BLEMidiServer.setNoteOffCallback(onNoteOff);

     #ifdef DEBUG_MIDI_TASK
    BLEMidiServer.enableDebugging();  // Comment to enable debugging NOT WORKING YET
    #endif
    
   
    midiTaskNOTES.Start(1); // Start the task on core 1 
  
}

void loop() {
    // put your main code here, to run repeatedly:
}

max22-, 2024-10-17:
why do you need to create a task ?

OSCmixer, 2024-10-17:
For better performance, maybe is problem that cannot disable debugging nimble messages...Normally iphone bluetooth to iphone works more smoothly also with fast notes...And trying to get best performance possible... Experimenting with esp32 options

max22-, 2024-10-17:
are you doing something else at the same time on the esp32 ?

OSCmixer, 2024-10-17:
YES this example https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-synthbasic3-audiokit/streams-synthbasic3-audiokit.ino