max22-/ESP32-BLE-MIDI #36: Cannot compile for esp32dev

edo44, 2024-12-10:
Arduino 2.3.4
ESP Arduino Core 3.0.7
Esp32Dev Module

```
In file included from c:\Users\ed\Documents\Arduino\libraries\ESP32-BLE-MIDI\src/BLEMidi.h:4,
                 from C:\Users\ed\AppData\Local\Temp\.arduinoIDE-unsaved20241110-23188-8vxbca.oo99l\01-Basic-Midi-Device\01-Basic-Midi-Device.ino:2:
c:\Users\ed\Documents\Arduino\libraries\ESP32-BLE-MIDI\src/utility/BLEMidiServer.h:18:10: error: 'void BLEMidiServerClass::onConnect(NimBLEServer*)' marked 'override', but does not override
   18 |     void onConnect(BLEServer* pServer) override;
      |          ^~~~~~~~~
c:\Users\ed\Documents\Arduino\libraries\ESP32-BLE-MIDI\src/utility/BLEMidiServer.h:19:10: error: 'void BLEMidiServerClass::onDisconnect(NimBLEServer*)' marked 'override', but does not override
   19 |     void onDisconnect(BLEServer* pServer) override;
      |          ^~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

Any help?

edo44, 2024-12-10:
Found the solution: had to update the NimBLE library from github.

max22-, 2024-12-11:
nice ! sorry for replying late

closed 2024-12-11 by max22-