max22-/ESP32-BLE-MIDI #41: Compile error with latest and NIMBle

dannyrubin, 2025-02-09:
I'm using the latest version of this library (0.3.2) to try and use BLEMidiServer.end().   I was previously using the NIMBLE-Arduino library, version 1.4.3 with no compile errors.  When I upgrade to any of the NIMBLE-Arduino versions 2.0 and higher I get the following compile errors (Arduino IDE version 1.8.19)

In file included from /Users/me/Documents/Arduino/libraries/ESP32-BLE-MIDI/src/BLEMidi.h:4,
                 from /Users/me/Documents/Arduino/BLEmidi-test/BLEmidi-test.ino:3:
/Users/me/Documents/Arduino/libraries/ESP32-BLE-MIDI/src/utility/BLEMidiServer.h:18:10: error: 'void BLEMidiServerClass::onConnect(NimBLEServer*)' marked 'override', but does not override
     void onConnect(BLEServer* pServer) override;
          ^~~~~~~~~
/Users/me/Documents/Arduino/libraries/ESP32-BLE-MIDI/src/utility/BLEMidiServer.h:19:10: error: 'void BLEMidiServerClass::onDisconnect(NimBLEServer*)' marked 'override', but does not override
     void onDisconnect(BLEServer* pServer) override;
          ^~~~~~~~~~~~
exit status 1
Error compiling for board Adafruit QT Py ESP32-C3.

Can anyone suggest a solution?
thanks

dannyrubin, 2025-02-18:
> Same here, I am going to try and download an older version.
>
> Any solution found yet?

not yet. let me know if you are successful. thanks

LeanderJDev, 2025-03-01:
I have the same problem. No solution yet though

ariknel, 2025-04-27:
HAve you solved it yet?

dannyrubin, 2025-04-28:
No progress on my side

ariknel, 2025-04-28:
> No progress on my sideOn Apr 27, 2025, at 12:26 PM, Arik ***@***.***&gt;

I have fixed it by downgrading arduinoBLE lib and nimBLE lib to versions
from november - september 2022 i believe it was. 4 at the end for nimBLE
something.. atleast way back to 2022 and. it's functioning as of now, works
with newest 3.2.0 esp idf

max22-, 2025-04-28:
I have worked on it this morning, i can send midi messages with the new NimBLE version, but i can't receive. I don't know why yet. i'll need to have another look at it

jblb, 2025-06-14:
Hi,
Version 1.4.3 of [NimBLE-Arduino](https://github.com/h2zero/NimBLE-Arduino) compile is fine, at least on basic-Midi-Device example, error appear only with version 2.X of NimBLE-Arduino. Someting to see [Migrating from 1.x to 2.x](https://github.com/h2zero/NimBLE-Arduino/blob/master/docs/1.x_to2.x_migration_guide.md)